bids-standard/pybv #3: Create a readthedocs page

choldgraf, 2018-11-20:
hey @chrisfilo can you create a readthedocs page for this repository? I don't think I have permissions to do so...

chrisfilo, 2018-11-20:
You should have all the permissions - could you try: https://readthedocs.org/dashboard/import/manual/?

choldgraf, 2018-11-20:
ah ok - I think it's up, closing now, thanks!

sappelhoff, 2018-11-23:
It's up but there is no content on it yet ... do we need to push our docs to it somehow?

choldgraf, 2018-11-23:
argg, the content is there but the build failed because we didn't install pybv as a part of the docs build. We need to add it to the docs requirements.txt file so that RTD can build with it.

@sappelhoff what's your readthedocs username? I'll add you as an admin

sappelhoff, 2018-11-23:
ahh okay, I don't have that much experience with readthedocs, but I'll learn more about it. My username there is `sappelhoff`.

choldgraf, 2018-11-23:
cool, just added you

sappelhoff, 2018-11-23:
was this line supposed to be the installation of pybv?
https://github.com/bids-standard/pybv/blob/bb49c1d8d905639562ee4943df0bace9f309c77c/.circleci/config.yml#L42

sappelhoff, 2018-12-15:
@choldgraf fyi ... the whole issue was fixed by ticking this box (see screenshot below) and adding our `requirements.txt` in the **Settings** of readthedocs ... the changes we did over CircleCI did not have any effect, because readthedocs creates its own environment instead of pulling already built docs over ...

docs are now testing AND building correctly ... and always referencing to "stable":

[image]