bids-standard/pybv #6: pip PyBV --> pybv

sappelhoff, 2018-11-23:
The current pip package `PyBV` should be renamed to `pybv` according to pep8 convention and to be consistent with `pyedf`, see: https://github.com/bids-standard/pyEDF/issues/5

Also, the contents of this repo and the `PyBV` package are not congruent, this is something to be solved soonish in order to avoid confusion.

choldgraf, 2018-11-23:
totally agree - we should do this whenever we make a pip release for pybv...who controls the pip page for pyBV?

sappelhoff, 2018-11-23:
Philip Alday :-)

sappelhoff, 2018-12-04:
thanks @palday!

@choldgraf, I have uploaded a test to `test.pypi`, see: https://test.pypi.org/project/pybv/

I have entered myself as the maintainer for now - if we do the real upload to pypi, should we create a shared bids-community account? Personally I would prefer if one of us takes the responsibility and hands over whenever a successor is needed and available

choldgraf, 2018-12-07:
@sappelhoff that works for me! Agree that it's good to keep things explicit, so long as it's clear that the tool is technically "owned" by the community, and that any specific maintainer has the right to step away from core maintenance privileges!

PR #7 (2018-12-10): [MRG] prepare for pushing to pypi
closes #6
closed 2018-12-10 by choldgraf