bids-standard/pybv #27: potential indexing issue

sappelhoff, 2019-07-18:
Could it be that we have an indexing issue in our code?

BrainVision uses 1-based indexing for writing the events, whereas in Python and MNE-Python, we use 0-based indexing.

Now imagine a user passing `events`: https://github.com/bids-standard/pybv/blob/cf8aa9d6904149d68ad9d9500315db5e785664a2/pybv/io.py#L46-L50

These events will most likely be 0-indexed.

However, later down in the code -when we write a VMRK file, we translate these indices one to one, disregarding that BrainVision used 1-based indexing. See: https://github.com/bids-standard/pybv/blob/cf8aa9d6904149d68ad9d9500315db5e785664a2/pybv/io.py#L163-L167

Later on in the testing, we actually see this issue: https://github.com/bids-standard/pybv/blob/cf8aa9d6904149d68ad9d9500315db5e785664a2/pybv/tests/test_bv_writer.py#L95-L96

note the comment. I may be wrong ... but it seems to me we should handle the indexing better

cc @tstenner @choldgraf

choldgraf, 2019-07-18:
ah this does seem like a potential problem - I wonder why we'd be testing for "within one" instead of "exactly the same" in the events...

sappelhoff, 2019-07-21:
MNE also documents that BrainVision is 1-indexed (not 0-indexed):

https://github.com/mne-tools/mne-python/blob/033f8c13fb03772c26142a1782c18a2a65dc9e8e/mne/io/brainvision/brainvision.py#L219

sappelhoff, 2019-07-22:
I got a reply from Brain Products:

```
Dear Stefan,

yes, this is correct, BrainVision indexing starts with 1, not with zero (comparable to Matlab indexing as opposed to Python). This means the index of the first data point is 1 and a marker on the first data point would have the position 1, accordingly.
Thank you for noticing that this is not documented in the specifications, I will pass on this information.

Best wishes,
Laura

```

PR #29 (2019-07-23): FIX+ENH: Fix indexing in events and add measurement date option to writer
This PR does two things

# 1
This PR exposes a `meas_date` argument to the writer function and a test suite. The default is `None` and maintains the current behavior.

# 2
This PR fixes #27 

I am adding `1` to each sample index, to account for BrainVision's 1-based indexing (whereas we expect an array that is 0-based indexed). Note, I changed the docstring for `events`. It now mentions that we expect a zero-based indexed array of events.

Currently the reasoning that BrainVision uses 1-based indexing is based on inspect [... cut]
closed 2019-07-23 by sappelhoff