bids-standard/pybv #28: fmt option no longer available for write_brainvision, probable update of README

alexrockhill, 2019-07-18:
In the README it says:

from pybv import write_brainvision

# for further parameters see our API documentation in the docs
write_brainvision(data, sfreq, ch_names, fname, tmpdir, events,
                  resolution=1e-6, fmt='binary_float32')

but in io.py the function definition for write_brainvision is

def write_brainvision(data, sfreq, ch_names, fname_base, folder_out,
                      events=None, resolution=1e-7, scale_data=True):
    """Write raw data to BrainVision format.

    Parameters
    ----------
    data : ndarray, shape (n_channels, n_times)
        The raw data to export. Data is assumed to be in
        **volts**. The data will be stored in **microvolts**.
    sfreq : int | float
        The sampling frequency of the data
    ch_names : list of strings, shape (n_channels,)
        The name of each channel
    fname_base : str
        The base name for the output files. Three files will be created
        (.vhdr, .vmrk, .eeg) and all will share this base name.
    folder_out : str
        The folder where output files will be saved.
    events : ndarray, shape (n_events, 2)
        Events to write in the marker file. This array has two columns.
        The first column is the index of each event (corresponding to the
        "time" dimension of the data array). The second column is a number
        associated with the "type" of event.
    resolution : float | ndarray
        The resolution **in volts** in which you'd like the data to be stored.
        By default, this will be 1e-7, or .1 microvolts. Since data is stored
        in microvolts, the data will be multiplied by the inverse of this
        factor, and all decimals will be cut off after this. So, this number
        controls the amount of round-trip resolution you want.
        This can be either a single float for all channels or an array with
        nchan elements.
    scale_data : bool
        Boolean indicating if the data is in volts and should be scaled to
        `resolution` (True), or if the data is already in the previously
        specified target resolution and should be left as-is (False).
        This is mostly useful if you have int16 data with a custom resolution.
    """

This doesn't include fmt

sappelhoff, 2019-07-19:
Could it be that you have an outdated version of `pybv`?

This is what the code says:

https://github.com/bids-standard/pybv/blob/cf8aa9d6904149d68ad9d9500315db5e785664a2/pybv/io.py#L27-L67

On a side note, you can format your issues using Markdown. Specifically, if you want to insert code blocks, you can use the ``` signs, as described in [this tutorial](https://help.github.com/en/articles/creating-and-highlighting-code-blocks)

sappelhoff, 2019-07-19:
You can install the development version using the following command: 

`pip install -U https://api.github.com/repos/bids-standard/pybv/zipball/master`

sappelhoff, 2019-07-23:
I just published version `0.1.0` on PyPI, so you should be able to do `pip install -U pybv` to get the most recent version :-)

closed 2019-07-23 by sappelhoff