bids-standard/pybv #30: Add support for annotations (including duration)

cbrnr, 2019-08-20:
It would be nice if `mne.Annotations` were supported in addition to events. Furthermore, to avoid loss of information the duration of each annotation needs to be stored, and I think the .vmrk file supports this right after the position (<Size in data points>, which is currently always set to 1). I suggest that the current `events` parameter should accept either an events array or an Annotations object and automatically do the right thing.

I'm happy to contribute a PR if you think this is a good idea.

sappelhoff, 2019-08-20:
Support for `mne.Annotations` would be nice indeed! A PR is very welcome.

> and I think the .vmrk file supports this right after the position (, which is currently always set to 1

Reading the BrainVision Core Data Format specification, it seems like this is the case. Although they ambiguously describe it as "length in data points", which could also refer to other information :thinking:  - to be on the safe side, we could inquire with Brain Products, whether this VMRK field truly refers to the duration of an event.

do you know whether MNE-Python in turn supports reading this kind of information when passed a BrainVision file?

cbrnr, 2019-08-20:
> to be on the safe side, we could inquire with Brain Products, whether this VMRK field truly refers to the duration of an event.

I don't know what other information this could refer to. I'm a bit hesitant to ask, because this could take a while. Plus, MNE already this this (see below).

> do you know whether MNE-Python in turn supports reading this kind of information when passed a BrainVision file?

Yes, when I read a BrainVision file I always get annotations with duration of 1 sample. When I manually edit a value, this is reflected in that annotation's duration.

sappelhoff, 2019-08-20:
> I don't know what other information this could refer to. I'm a bit hesitant to ask, because this could take a while. Plus, MNE already this this (see below).

okay, fair enough to go ahead. Perhaps you could still inquire with them - but I won't block this if you don't :+1:

cbrnr, 2019-08-20:
Maybe @dmedine can weigh in?

cbrnr, 2019-08-21:
I think we want to avoid adding MNE as a dependency, so we should support a duration. What about accepting an (N, 3) ndarray where the middle column contains the durations? Ideally, this should also work with a list of lists, because we might want to supply string values for the types.

tstenner, 2019-08-21:
I could check an Analyzer exported file (bad segment markers map to events), but IIRC the duration is indeed the event duration.

PR #33 (2019-08-22): Support writing custom event durations
Fixes #30. I have written some custom functions to convert `mne.Annotations` into a suitable (n, 3) ndarray - the question is where should I include these functions? Maybe the best place would be directly in MNE. Here's the code:

```
def annotations_to_array(annotations, fs):
    array = np.zeros((len(annotations), 3), dtype=int)
    ids = _unique_ids(annotations.description)
    for row, annotation in enumerate(annotations):
        array[row, 0] = annotation["onset"] * fs
        array[row, 1] = ids[annotation["description"]]
        array[row, 2] = an [... cut]
closed 2019-08-22 by sappelhoff