bids-standard/pybv #31: Exported file contains additional event

cbrnr, 2019-08-20:
When exporting a file with events, the resulting .vmrk file has one additional event with the measurement date. This is not ideal because when I read the file back in, I suddenly have one additional event as compared to the original data. Isn't there another way to store the measurement date? If not, could we add an option to disable storing the additional event (e.g. use `meas_date=None` for this purpose instead of writing a zero date, or define a new value such as `meas_date="none"`)?

sappelhoff, 2019-08-20:
just to be sure, you are talking about this line, aren't you? https://github.com/bids-standard/pybv/blob/d6e9ff92424e6480dd36a611c5f35352c79474d9/pybv/io.py#L176

you are right ... we are adding an event that wasn't there before ... thanks for the catch.

I would be fine with defining a new value `'skip'` for the `meas_date` parameter in `write_brainvision` to skip writing that additional event.

cbrnr, 2019-08-20:
Yes, this is the line. I can make a PR if you want.

sappelhoff, 2019-08-20:
PR welcome, and very cool to see you want to add this to MNE-LAB :-)

cbrnr, 2019-08-20:
I'll open a PR soon! PS: I've already merged the changes, so MNELAB can now export to BrainVision ❤️ !

cbrnr, 2019-08-21:
Could we use `meas_date=None` for this purpose? I don't think mapping this to the date `'00000000000000000000'` has any benefits for the user, because an additional event with an arbitrary date is added.

PR #32 (2019-08-21): Add option to disable writing a meas_date event
Fixes #31.
closed 2019-08-21 by sappelhoff