bids-standard/pybv #34: New release on PyPI?

cbrnr, 2019-08-26:
Now that #32 and #33 are merged it would be really great if you could cut a new release - these features are really important for correctly exporting annotations in MNELAB.

sappelhoff, 2019-08-26:
@cbrnr given that you are a trusted member of MNE as well, would you mind doing the release yourself? I can grant you the necessary permissions.

sappelhoff, 2019-08-26:
and see https://github.com/bids-standard/pybv/blob/master/CONTRIBUTING.md "making a release" section

cbrnr, 2019-08-26:
Sure, I'll do it!

sappelhoff, 2019-08-26:
Cool, you should have permissions on GitHub and PyPi now. Please let me know if there is something about the release guidelines that should be improved!

cbrnr, 2019-08-26:
OK, will do! Just a quick question, the current master already has version 0.3.0dev, which means that the version I am going to push will be 0.3.0. What if I just wanted to push a minor update such as 0.2.1? How would this work? MNE does it with backports, but this always seemed a lot of overhead to me.

sappelhoff, 2019-08-26:
We are setting the `dev` version to the next "expected released" ... that was `0.3.0`, and hence the current version is `0.3.0.dev0` ... now it turns out that we do a `0.2.1` in between, so basically you change to `0.2.1` for now (in your release commit), and then after the release back to e.g., `0.3.0.dev1`

Not sure whether this is the canonical way to do things - but I feel like the versions make sense.

cbrnr, 2019-08-26:
OK, this sounds like a good plan (I was interested how to handle exactly this problem in MNELAB)!

So which version should I release now? 0.2.1 or 0.3.0?

sappelhoff, 2019-08-26:
> OK, this sounds like a good plan (I was interested how to handle exactly this problem in 
MNELAB)!

cool :-)

For this we go according to semantic versioning ... if we have a new feature, do a `minor` (=x.X.x) ... if it's mostly bugfixes / streamlining, we go with `patch` (=x.x.X) ... if it's not backward compatible, go with `major` (=X.x.x)

cbrnr, 2019-08-26:
👍 so with two new features I guess 0.3.0 it is then.

sappelhoff, 2019-08-26:
+1 from my side for that :-)

cbrnr, 2019-08-26:
@sappelhoff did you also add me to test-pypi? I get an error when trying to upload, and I don't want to skip this step.

sappelhoff, 2019-08-26:
done now! Completely forgot about it earlier. thanks for reminding.

cbrnr, 2019-08-26:
Alright, stuck in step 16 - I can't `git push --follow-tags upstream`, presumably because I've enabled 2FA and I normally use SSH, but upstream is `https://github.com/bids-standard/pybv.git`:

```
remote: Invalid username or password.
```

Can/should I just change `upstream` to SSH, i.e. `git@github.com:bids-standard/pybv.git`? If so, maybe this could be added to the guide.

cbrnr, 2019-08-26:
PS: Is this step necessary? I've always created the tags directly with the release on the GitHub website.

sappelhoff, 2019-08-26:
As long as you push the annotated tag to master, it's fine - choose a method that works!

> PS: Is this step necessary? I've always created the tags directly with the release on the GitHub website.

Making a release from GitHub only does not create an annotated tag ... but we may want annotated tags for releases: https://stackoverflow.com/a/25996877/5201771 rather than normal tags (which seem to be more for local work)

cbrnr, 2019-08-26:
OK, changing the remote URL worked. I didn't know that releases should have annotated tags, so far I thought that as long as `git checkout v0.2.0` works (which works with lightweight tags) we're good to go. But I'm sure there's a lot I don't know about tags so I'll just proceed as per the instructions 😄.

cbrnr, 2019-08-26:
OK done - 0.2.0 has been released 🎉!

closed 2019-08-26 by cbrnr

sappelhoff, 2019-08-26:
Thanks @cbrnr :-)