bids-standard/pybv #38: Set unit to microvolts

cbrnr, 2020-03-30:
It seems like it is currently not possible to set the resolution and unit independently. As soon as `resolution <= -7` (which is `True` for the default), the unit is set to nV.

However, I just had a conversation with a BrainProducts developer regarding potential issues with the unit. Specifically, he said that BrainVision Analyzer currently expects the unit to be µV. If it's something else, auto-scaling doesn't work (at least the initial view).

Therefore, I would like to suggest to add an option that controls the unit. Possible values could include `'V'`, `'mV'`, `'uV'`, `'µV'`, and `'nV'`. I would set the default to `'uV'` (which is also the expected unit if the unit field in the exported file is empty).

cbrnr, 2020-03-30:
There could also be an `'auto'` value, which would replicate the current behavior.

sappelhoff, 2020-03-30:
Hi @cbrnr ... always good to get some info from Brain Products. Thanks for reporting the potential issue :-)

I think your suggestion is reasonable, especially when `'auto'` is added as an option.
                       

@tstenner implemented that current behavior, perhaps he has something to say about this.

tstenner, 2020-03-30:
The PRs in question (#17 and #20) will have their anniversary in less than a month so I don't recall the exact reasoning, but I think giving the scale in µV uses more digits for the same precision or cuts off more digits than I was comfortable with.

PR #39 (2020-03-31): ENH: Add unit parameter to write_brainvision
Fixes #38.

Test with the following code snippet (this could be included as a proper test):

```python
from pathlib import Path
import numpy as np
import mne
from mne.io import read_raw_brainvision
import pybv

rng = np.random.default_rng(42)
data = rng.normal(loc=0, scale=1e-5, size=(10, 10000))  # typical EEG scale
info = mne.create_info(10, 256, "eeg")
raw1 = mne.io.RawArray(data, info)

fpath = "."
fname = "export"
resolution = 1e-7
unit = "µV"  # "V", "mV", "uV", "µV", "nV", "auto"

pybv.write_brainvision(data=raw1.get_data(),  # always in V
   [... cut]
closed 2020-03-31 by sappelhoff