bids-standard/pybv #44: Let pybv write [coordinates] section to vhdr files

sappelhoff, 2020-07-19:
See BrainVision data file [specification](https://www.brainproducts.com/productdetails.php?id=21&tab=5).

The OPTIONAL `[Coordinates]` section in a `vhdr` file comes right after the REQUIRED `[Channels]` section, and it looks like this:

```Text
[Coordinates]
;Each entry: Ch<Channel number>=<Radius>,<Theta>,<Phi>
Ch1=1,-60,-51
Ch2=1,-49,-29
Ch3=1,-45,0
Ch4=1,-49,29
```

For the radius, the spec say the following [**emphasis** by me]:

>“The radius r **specifies the distance (in millimeters)** between point P and the origin of the coordinate system. The only **exceptions are r = 0 and r = 1**. **r = 0 signifies an invalid position**, for instance when the position of an electrode is not known. When realistic electrode coordinates are used, r can have a different value for each channel. **In other cases, the value of r should be the same for all the channels if a spherical head model is used**. For instance, in the Analyzer's standard coordinate system, r = 1”.

All coordinates that pybv writes to the `[Coordinates]` section must first be transformed to the *Analyzer's standard coordinate system*, which happens to be the `Captrak` system (more info in [FieldTrip wiki](http://www.fieldtriptoolbox.org/faq/how_are_the_different_head_and_mri_coordinate_systems_defined/#details-of-the-captrak-coordinate-system)). For this conversion to work, we will need the Nasion, left, and right preauricular point coordinates (to rescale the coordsystem).

I suggest that furthermore, pybv writes a more extensive comment in the `[Coordinates]` section, i.d.:

```Text
[Coordinates]
; Each entry: Ch<Channel number>=<Radius>,<Theta>,<Phi>
; Radius defined in millimeters (mm).
; Radius of 0 signifies invalid position. Radius of 1 signifies a spherical head model.
; The coordinate system is "Captrak", see definition of axes below.
; The X-axis goes from the left preauricular point towards (through) the right preauricular point.
; The Y-axis goes orthogonally to the X-axis and towards (through) the nasion.
; The Z-axis goes orthogonally to the X-Y-plane upwards.
Ch1=1,-60,-51
Ch2=1,-49,-29
Ch3=1,-45,0
Ch4=1,-49,29
```

sappelhoff, 2021-04-03:
this will be particularly interesting once MNE starts to export to pybv ... because we'll want to capture as much of the FIF data as possible for a round-trip.

x-ref: https://github.com/mne-tools/mne-python/issues/8060#issuecomment-812645784

sappelhoff, 2021-08-08:
as discussed in #77, the input for a `coordinates` parameter in `write_brainvision` could be a list of dicts:

```
[
    {
        "name": "LPA",
        "x": 0.3,
        "y": 4.5,
        "z": 6.1,
    },
    {
        "name": "Fpz",
        "radius": 3.1,
        "theta": 22,
        "phi": 33,
    }
]
```

pybv would then:

- check for "LPA", "RPA", and "Nasion", and write them as "comment" (because VHDR only supports existing data channels in their coordinate section; see [my example](https://github.com/bids-standard/pybv/issues/44#issue-660664875))
- check for all other "names" and map them to channel indices as present in the data (we write indices, not names to VHDR)
- raise a warning if there are more or less names than in the data
- if less names, the missing names are entered with zeros (that is the "NaN" representation in VHDR)
- if more names, the superfluous names are just ignored

PS: I have a very lengthy email discussion on this topic with Brain Products that I didn't post here ... But I think what we plan here would be spec conformant: Especially if we put coordsystem info, units, and landmarks as "comments".

BV Readers can then optionally check the VHDR for whether the file was created by "pybv" and if it was, make use of that "comment".

BV Readers that *dont* check for that, will still get sensible data: At least as sensible as the underspecified "Coordinates" section allows :-)

hoechenberger, 2021-08-09:
This sounds great!

sappelhoff, 2022-05-26:
we'd probably need to look at these three functions in mne

- get_ras_to_neuromag_trans
- compute_native_head_t
- transform_to_head

and simplify them for use here without adding mne as a dependency.