bids-standard/pybv #45: Roundtrip fails for data in "short" format when writing to "binary_int16"

sappelhoff, 2020-11-05:
first discovered in https://github.com/mne-tools/mne-bids/pull/610

@hoechenberger prepared a MWE to reproduce the issue:

```Python
# %%
from pathlib import Path
import numpy as np
import pybv
import mne

raw_fname = (Path(mne.__file__).parent / 'io' / 'brainvision' / 'tests' /
             'data' / 'test.vhdr')
raw = mne.io.read_raw_brainvision(raw_fname, preload=True)
assert raw.orig_format == 'short'

pybv.write_brainvision(data=raw.get_data(),
                       sfreq=raw.info['sfreq'],
                       ch_names=raw.ch_names,
                       fname_base='test',
                       folder_out='/tmp',
                       fmt='binary_int16')

raw_read = mne.io.read_raw_brainvision('/tmp/test.vhdr', preload=True)
assert np.allclose(raw.get_data(), raw_read.get_data())
```

--> fails.

Assertion holds if using fmt='binary_float32', although it should be totally fine to write short data as binary_int16

sappelhoff, 2020-11-05:
I've been working on this for quite a bit now and one thing that's fishy for sure is, that if the BV file contains channels that have (i) differently scaled units in Volts, or (ii) have channels that are not measured in Volts, then we have a problem.

cc @hoechenberger just to keep you informed

sappelhoff, 2020-11-05:
also, when writing with `resolution=0.5` and `unit='µV'` I get a resolution of 500000 in the written file :angel:  (screenshot:)

[image]

hoechenberger, 2020-11-05:
> also, when writing with `resolution=0.5` and `unit='µV'` I get a resolution of 500000 in the written file 👼 (screenshot:)

Wow this is clearly wrong…
We should also adjust the `unit` and `resolution` keywords to accept an interable of length `n_chans`

sappelhoff, 2020-11-05:
> Wow this is clearly wrong…

agreed: #48 #49 

>  We should also adjust the unit and resolution keywords to accept an interable of length n_chans

agreed: #50

sappelhoff, 2020-11-05:
when it rains it pours ... I found yet another issue if you pass channel names that contain a comma. And I sent you an invite to join this repo with "write" privileges, Richard.

hoechenberger, 2020-11-05:
> And I sent you an invite to join this repo with "write" privileges, Richard.

(How can I get outta here without exposing myself as a lazy lurker!)
((Probably too late now anyway))
(((...argh...)))

sappelhoff, 2020-11-06:
the issue can also nicely be replicated by using this test:

https://github.com/bids-standard/pybv/blob/54da125d09b3b3fb83178e7fe06996929aa1ef3d/pybv/tests/test_bv_writer.py#L175-L185

and making this small change to the first lines:

```diff
+@pytest.mark.parametrize("format", ("binary_float32", "binary_int16")) 
 @pytest.mark.parametrize("resolution", np.logspace(-1, -9, 9)) 
 @pytest.mark.parametrize("unit", ["V", "mV", "uV", "µV", "nV", None]) 
+ def test_unit_resolution(format, resolution, unit): 
     """Test different combinations of units and resolutions.""" 
     tmpdir = _mktmpdir() 
     write_brainvision(data, sfreq, ch_names, fname, tmpdir, 
+                       resolution=resolution, unit=unit, fmt=format) 
```

sappelhoff, 2020-11-07:
relevant comment, see: https://github.com/bids-standard/pybv/pull/58#issuecomment-723446686

> sometimes we scale the data so high, that it cannot be represented in int16 anymore ... whereas it's totally possible in float32.

hoechenberger, 2020-11-08:
> And I sent you an invite to join this repo with "write" privileges, Richard.

I don't know, maybe I missed it because I've received so many notifications of GH stuff happening, but long story short: I unfortunately cannot find an invitation, neither in my mailbox, nor in the GH notifications. But both of them I clear ~daily so I may have accidentally overlooked & removed it..?

sappelhoff, 2020-11-08:
you accepted it apparently :) 

[image]

hoechenberger, 2020-11-08:
> you accepted it apparently :)

It's a little bit unsettling if you obviously lose track of what you're doing… 🤪

cbrnr, 2020-11-08:
BTW what does "Outside Collaborator" mean?

sappelhoff, 2020-11-08:
> BTW what does "Outside Collaborator" mean?

just that you are not part of the bids-standard organization on GitHub :-)

PR #62 (2020-11-08): FIX: do not write data if it cannot be represented in fmt, given unit and resolution
fixes #45 

previously we allowed users to write their data in whatever combination of fmt, unit, and resolution they wanted. --> in some cases this lead to data not being accurately represented ... for example, high resolution µV data in  binary_int16 format.

In this PR I add a check to see if the data *can* be written as desired ... and if not, I raise a ValueError.
closed 2020-11-08 by sappelhoff