bids-standard/pybv #52: float sampling frequencies are not supported

sappelhoff, 2020-11-05:
the BrainVision format specifies the sampling frequency of the data as "sampling interval in microseconds"

We are converting the sampling frequency in Hertz like this:

https://github.com/bids-standard/pybv/blob/76e8bea84d6594ca8e87f8fe367896167e43c641/pybv/io.py#L257

This means that we do not currently support sampling frequencies like `11.5`  ... I am not even sure whether that can reasonably exists, but we should add a check for that in the code anyways.

hoechenberger, 2020-11-05:
Maybe we should contact Brain Products and inquire if it's even intended to be a float?

cbrnr, 2020-11-06:
I'm not sure if any amplifier supports non-integer sampling frequencies, but physically there's no reason why it can't be a float. In fact, most clocks will not be 100% accurate so that you will end up with e.g. 512.01 Hz instead of the advertised 512 Hz if you measure it with a precise clock. That's what happens in XDF by default, and our effective (calculated) sampling frequencies are float in general (also due to the limited time range we have to compute that number).

So what I'm trying to say is we should allow sampling rates to be a float, and this should probably also be the default.

hoechenberger, 2020-11-06:
But we're talking about the duration of one **sampling cycle** in nano-seconds, not the **sampling frequency** per se.

I guess I'm just worried that if we start writing floats, other apps like BV Analyzer may not be able to read those files anymore. I don't have access to Analyzer so cannot test.

cbrnr, 2020-11-06:
Then I don't understand what the problem is. With duration in ns we can represent non-integer sampling frequencies, right?

hoechenberger, 2020-11-06:
Yes! But as I understood @sappelhoff's first comment, BV doesn't ask us to write the **sampling frequency**, but the **duration** of a sampling cycle in µs (sorry, I accidentally wrote ns before). So for sfreq==500 Hz, we'd write 2000, because the duration of one sampling cycle is 2000 µs.

hoechenberger, 2020-11-06:
If sreq were 512 Hz, the duration would be 1953.125 µs, and I'm not sure if that's valid or not :)

cbrnr, 2020-11-06:
OK, got it. But in this case it is even more important to support floats, otherwise we can't represent the most commonly used sampling frequencies. We could try to write ints if the result is close enough (e.g. 2000.000 would be 2000), but then the question is what criterion do we use? How many decimal places to we assume significant?

sappelhoff, 2020-11-06:
we *can* write the true sampling frequency in the [comment] section at least, see: https://github.com/mne-tools/mne-python/blob/28b36c06a744074b54993128fa1908ea10464ff2/mne/io/brainvision/tests/data/test.vhdr#L56-L62

> But in this case it is even more important to support floats, otherwise we can't represent the most commonly used sampling frequencies. 

agreed ... I really wonder why brainvision decided to represent sfreq as the "sampling interval in microseconds ..."

cbrnr, 2020-11-06:
> agreed ... I really wonder why brainvision decided to represent sfreq as the "sampling interval in microseconds ..."

Probably because their amps only use sampling intervals with integer durations when displayed in microseconds 😄 &ndash; I don't think the format was intended to be used by everyone. But it's nice that we can make it work.

sappelhoff, 2020-11-06:
lucky for us, the spec allows floats, at least that's how I interpret this part, see screenshot:

[image]

sappelhoff, 2020-11-06:
@cbrnr do you want to prepare a fix?

cbrnr, 2020-11-06:
Yes, will do!

PR #59 (2020-11-07): Sampling interval can be float
Fixes #52. I've also introduced f-strings in this block of code - do you want me to convert the rest as well?
closed 2020-11-07 by sappelhoff