bids-standard/pybv #69: Simplify Makefile rule?

cbrnr, 2021-01-03:
I'm not a Makefile expert, but could the rule

```
pep:
	@$(MAKE) -k flake pydocstyle check-manifest
```

be simplified to

```
pep: flake pydocstyle check-manifest
```

? If not I'd be curious to learn why.

Furthermore, I think it is common practice to define phony targets explicitly, i.e. the first rule should be

```
.PHONY: all inplace test flake pydocstyle check-manifest pep build-doc
```

sappelhoff, 2021-01-03:
Regarding your first point: I don't know, @adam2392 added the file, and I only slightly edited it. Please feel free to adjust the makefile to follow best practice. 

Regarding your second point, I think you are right! See also https://stackoverflow.com/a/2145605

adam2392, 2021-01-04:
Yeah I copy pasted from mne-bids to help me test docs when making the pybv PR, so I have no idea what's best practice :p

PR #70 (2021-01-04): Improve Makefile
Fixes #69. I tested `make pep` on my local machine and it is still working.
closed 2021-01-04 by sappelhoff