bids-standard/pybv #72: The micro sign

hoechenberger, 2021-01-27:
So… I was recently browsing Wikipedia, reading through some unrelated issues, when I suddenly happened to come across the [`Micro-`](https://en.wikipedia.org/wiki/Micro-) lemma.

And, what do you know, it starts off with this sentence (highlights by me)

> Micro- (**Greek letter μ** or **legacy micro symbol µ**) is a unit prefix in the metric system …

In the following paragraph, it is stated:

> The official symbol for the SI prefix micro- is a Greek lowercase mu (μ). (…) According to The Unicode Consortium, the Greek letter character is preferred,[6] but implementations must recognize the micro sign as well.

So to me this means that if anything, we should use the Greek letter and convert to the Greek letter whenever we encounter the "micro" sign… Or are we using the legacy micro sign because that's what the BrainVision software uses?

cbrnr, 2021-01-27:
I know, we should really be using the Greek mu because that's what they recommend. I think we're using the micro sign because that's what BrainProducts specifies.

sappelhoff, 2021-01-27:
> Or are we using the legacy micro sign because that's what the BrainVision software uses?

:point_up: this is the reason, indeed. It says so in the BrainVision spec. 

See also https://github.com/mne-tools/mne-python/pull/7497#issuecomment-603494677 which I found funny at the time

closed 2021-01-27 by sappelhoff