bids-standard/pybv #73: ALERT: Potential exposure of PyPI tokens

effigies, 2021-04-15:
This morning (15 April 2021), I received a security update from CodeCov: https://about.codecov.io/security-update/

PyBV was affected by this issue because we use the CodeCov GitHub action, which means tokens were potentially exposed. There are PYPI secrets in the repository settings that should be revoked.

I strongly suggest moving to API tokens that are more easily rotated in the future.

sappelhoff, 2021-04-15:
Thanks @effigies, I changed pypi passwords and updated the tokens.

> I strongly suggest moving to API tokens that are more easily rotated in the future.

do you have further information on what you mean by that?

effigies, 2021-04-15:
https://pypi.org/help/#apitoken

sappelhoff, 2021-04-15:
I see, thank you.

I generated tokens with repo-specific upload permissions and updated the secrets in GitHub accordingly.

closed 2021-04-15 by sappelhoff