bids-standard/pybv #79: Tests fail with little-endian machines

Aniket-Pradhan, 2021-09-13:
Hello Devs

Thanks a lot for providing pybv, it is a very useful tool that is used in neuroimaging. I am trying to include python-pybv for Fedora, a Linux distribution. I was trying to run the tests, but they fail on a big-endian machine, s390x architecture to be precise. Below is a snippet of a failing test:

```bash
___________________ test_comma_in_ch_name[ch_names_tricky1] ____________________
tmpdir = local('/tmp/pytest-of-mockbuild/pytest-0/test_comma_in_ch_name_ch_names1')
ch_names_tricky = ['ch_0,foo', 'ch_1,foo', 'ch_2,foo', 'ch_3,foo', 'ch_4,foo', 'ch_5,foo', ...]
    @requires_version("mne", min_version="0.22")
    @pytest.mark.parametrize("ch_names_tricky",
                             [[ch + ' f o o' for ch in ch_names],
                              [ch + ',foo' for ch in ch_names],
                              ]
                             )
    def test_comma_in_ch_name(tmpdir, ch_names_tricky):
        """Test that writing channel names with special characters works."""
        # write and read data to BV format
        write_brainvision(data=data, sfreq=sfreq, ch_names=ch_names_tricky,
                          fname_base=fname, folder_out=tmpdir)
        vhdr_fname = tmpdir / fname + '.vhdr'
        raw_written = mne.io.read_raw_brainvision(vhdr_fname=vhdr_fname,
                                                  preload=True)
    
        assert ch_names_tricky == raw_written.ch_names  # channels
    
>       assert_allclose(data, raw_written._data)  # data round-trip
E       AssertionError: 
E       Not equal to tolerance rtol=1e-07, atol=0
E       
E       x and y nan location mismatch:
E        x: array([[-7.031873e-06, -4.902824e-06, -3.218143e-06, ..., -4.052640e-06,
E               -5.178996e-06,  1.410489e-06],
E              [ 1.635612e-05,  1.622443e-05,  4.016866e-06, ..., -5.808766e-06,...
E        y: array([[ 4.759927e-16, -4.722928e+18, -6.062177e-01, ...,  4.668447e-41,
E               -8.139047e+19, -1.879789e-24],
E              [-1.627288e-18, -5.858951e-38, -1.282439e-13, ..., -3.478547e-04,...
pybv/tests/test_bv_writer.py:147: AssertionError
```

The tests work fine on the little-endian machines, which basically implies that the bug might be where/how we are storing the files. I tried going through pybv's io and from what I could understand it is writing the file in the UTF-8 format, therefore the endianness [shouldn't matter](https://stackoverflow.com/a/23831655).

However, when I went through mne's brainvision io, I could see that they [open the file as a binary file](https://github.com/mne-tools/mne-python/blob/main/mne/io/brainvision/brainvision.py#L75) which could be causing the tests to fail.

I am not sure if this is the case, but if it is I would also file a ticket at mne so that we could get a resolution and support more architectures. If you could look into this and let us know it would be great!

sappelhoff, 2021-09-14:
Hi @Aniket-Pradhan, thanks a lot for reporting this issue. Unfortunately I have no idea about endianness and how to deal with it for pybv - but I'll take some time to look into it in two weeks. Perhaps someone else like @cbrnr or @hoechenberger  or possibly @tstenner can help you in the meantime!

cbrnr, 2021-09-14:
We are currently writing data using `numpy.ndarray.tofile()`, which ignores endianness (https://numpy.org/doc/stable/reference/generated/numpy.ndarray.tofile.html). So unless someone adds explicit support for this (meaning we'd have to roll our own array writer), I'm afraid there's nothing much we can do at this point. But maybe this is not that difficult and it works by detecting the endianness of the current machine and doing some `byteswap()` calls?

sappelhoff, 2021-09-14:
> We are currently writing data using numpy.ndarray.tofile(), which ignores endianness 

Does that mean that brainvision files written on big endian machines can only be accurately read on big endian machines, and vice versa for little endian?

cbrnr, 2021-09-14:
> Does that mean that brainvision files written on big endian machines can only be accurately read on big endian machines, and vice versa for little endian?

Looks like it, yes.

Aniket-Pradhan, 2021-09-14:
Hey @cbrnr and @sappelhoff .

I had ran the test on a big-endian machine itself, so I was not trying to read/write files on different endian machines. Although yes that could be a scope for development, as pointed out by @cbrnr .

Since the files are being written on a big-endian machine I am assuming it should be read as it is, however since the test makes use of [`mne.io.read_raw_brainvision`](https://github.com/mne-tools/mne-python/blob/main/mne/io/brainvision/brainvision.py#L75), as I had pointed out it ignores the endianness when reading the file. I was not sure of it, thats why I had raised the ticket here to confirm the same.

cbrnr, 2021-09-14:
Yes, it's surprising that the test fails given that both writer and reader are on the same big-endian machine. However, the real issue here is that I actually don't know what the byte order should be according to the standard. In fact, the specification does not mention endianness. @sappelhoff do you have a suggestion on how we should proceed? Now that ARM64 is getting more popular with M1 Macs this issue will definitely haunt us again.

sappelhoff, 2021-09-14:
> I actually don't know what the byte order should be according to the standard. In fact, the specification does not mention endianness. @sappelhoff do you have a suggestion on how we should proceed?

I think an Email to Brain Products is warranted to find out about their thoughts on byte order in BrainVision files. I'll draft one and share the correspondence here. Maybe they have some currently not communicated rule like: "all BrainVision files are little endian" (that would be the lucky case for us, I think).

But first of all, I need to understand the problem a little better:

1. we use `np.ndarray.tofile`, which says: "Information on endianness and precision is lost, so this method is not a good choice for files intended to archive data or transport data between machines with different endianness."
1. How can I understand that? --> is information about endianness encoded in some other binary formats, and in contrast, `.tofile` just starts writing bytes, without writing the endianness to file?
1. if my understanding is correct - why is an error raised, when data is written on a big-endian machine (in big-endian, because `.tofile` does not change endianness), and then read with `mne.io.read_raw_brainvision` (which ignores endianness, so reads as big-endian on a big-endian machine, and as small-endian ond a small-endian machine)

I feel something in my point 3 is off, but what?

Are Numpy arrays not always in the same endian mode as the machine they are on?

cbrnr, 2021-09-14:
I think you are correct. Some file formats store byte order with a [byte order mark (BOM)](https://en.wikipedia.org/wiki/Byte_order_mark), but this is not the case with files written by `numpy.ndarray.tofile()`. This function just uses the byte order of the host.

Before contacting BrainProducts, I would also like to find out why the error is occurring in the first place. I also thought that within a given platform, writing and reading should work and problems should only appear when crossing endianness borders (e.g. reading a little-endian file on a big-endian machine).

For that, we'd need a little more detail on the exact error. For example, there is also a `nan location mismatch` – I'd like to know if this is the only error or if other non-nan values are also different. Maybe we should start with a minimal example (creating an array, writing with `tofile`, and reading with `fromfile` with both byte orders), but I don't have access to a big-endian machine. Are these available on one of the CI hosters?

sappelhoff, 2021-09-14:
It seems to be possible for `s390x` even when using Travis: https://blog.travis-ci.com/2019-11-12-multi-cpu-architecture-ibm-power-ibm-z

(based on 10 minute search, so there may be more)

cbrnr, 2021-09-14:
Time to set up that CI then I guess? Alternatively, we could decide to support only little endian (after all big endian machines are really uncommon these days). This might be a good solution if BrainProducts tells us that their format requires little endian. But even in that case it would be nice to make reading and writing work on big endian architectures by forcing little endian when reading and writing.

sappelhoff, 2021-09-14:
yes, let's:

1. set up Travis and reproduce the error
2. make sure we understand everything
3. contact Brain Products and ask them what they think

With the potential solution to only write little-endian (which might be what BP tells us anyhow)

I'll get to 1. only in 2 weeks or so. If somebody else wants to give it a shot, please go ahead.

cbrnr, 2021-09-14:
Maybe this even works with GitHub Actions: https://github.com/marketplace/actions/run-on-architecture

Aniket-Pradhan, 2021-09-15:
Hey, sorry I got a bit late.

I have configured a workflow in a fork of this repository [here](https://github.com/Aniket-Pradhan/pybv/blob/main/.github/workflows/run-on-arch-workflow.yml).

[Here](https://github.com/Aniket-Pradhan/pybv/actions/runs/1236913105) are the failing workflows for s390x and ppc64le architectures. I am not sure as to why ppc64le is failing because it is a little-endian CPU arch (although it is derived from its big-endian architecture).

[Here](https://kojipkgs.fedoraproject.org//work/tasks/3748/75613748/build.log) are the initial logs on which I submitted the initial ticket.

The fact that ppc64le is also failing the tests makes the issue more confusing, and makes me wonder if the endianness is actually the issue or not.

**P.S:** About the workflow, I had to use the distro's package for scipy because there are no wheels available for these architectures on Pypi and pip tries to manually build it. It takes [~3.5 hours to build scipy from source and then test pybv on the s390x arch](https://github.com/Aniket-Pradhan/pybv/runs/3606219047?check_suite_focus=true). I tried to use pip for all other dependencies so there's no version conflict now or in the future.

sappelhoff, 2021-09-16:
Cool, thanks! 

So we replicate the issue, but also see another unexpected but probably related issue.

>  I am not sure as to why ppc64le is failing because it is a little-endian CPU arch (although it is derived from its big-endian architecture).

Could you clarify what you mean by that? I thought endianness is given by the cpu architecture, so what does "derived from big endian arch" mean?

> P.S: About the workflow, I had to use the distro's package for scipy because there are no wheels available for these architectures on Pypi and pip tries to manually build it. It takes ~3.5 hours to [...]

That's now problem at all, neither pybv, nor `mne.io.brainvision` use scipy in any way. We just need it because it's a dependency to install MNE-Python.

> EDIT: I can help with debugging in case you don't have access to a ppc64le machine. I have ssh access to one and I can use it for standard debugging if needed. For now, I am trying to write a file which we can verify on some other system if pybv is working as expected or not.

Your help would be greatly appreciated, we have a high interest to fix this issue ... or to at least understand it first and then discuss possible fixes.

I'll write an email to BrainVision today about byte order in their files. Perhaps that'll already clear up some things.

Aniket-Pradhan, 2021-09-16:
> So we replicate the issue, but also see another unexpected but probably related issue.

The failure on the ppc64le architecture seems to be an issue with docker (and the chroot environment I was trying with earlier). When I tried the tests on the qemu vm, the tests were passing successfully. I also set up a [travis job](https://app.travis-ci.com/github/Aniket-Pradhan/pybv) to verify the same.

I am completely blank on why the tests do not work with docker/chroot. I'll try to ping some other people who might know more about this.

The only issue we have now is the s390x failure which is also replicated in the travis job.

> > I am not sure as to why ppc64le is failing because it is a little-endian CPU arch (although it is derived from its big-endian architecture).
> 
> Could you clarify what you mean by that? I thought endianness is given by the cpu architecture, so what does "derived from big endian arch" mean?

The ppc64 is a big-endian architecture. The le version is little-endian. I thought since the base is a big-endian arch maybe that could have been affecting the build. But that was a dumb assumption, and seems to be invalidated from the travis test.

sappelhoff, 2021-09-21:
short update: I've been in touch with BP support, and they are now looking into it and will get back to me. I'll let you know.

sappelhoff, 2021-09-21:
That was quick, I got this reply:

> Thanks again for reaching out about this question.
I had the chance to talk to our developers and they confirmed that the binary eeg file is always following the little-endian order without using byte order mark (BOM).
I will point out to our PM team that the specification is currently lacking this detail.

cbrnr, 2021-09-21:
OK, that was expected. So now we need to decide whether we want to support big endian architectures or not, both for pybv and mne.

Aniket-Pradhan, 2021-09-21:
> OK, that was expected. So now we need to decide whether we want to support big endian architectures or not, both for pybv and mne.

Unless a new specification comes, which includes a BOM, I guess we could make little modifications to `pybv` so that it writes/encodes the eeg file in little-endian irrespective of the machine its running on. Same for `mne`, although, from my assumption its already trying to read in little-endian format, so it wouldn't require much changes.

cbrnr, 2021-09-21:
I'm not sure if MNE reads in little endian. It looks like at least [this line](https://github.com/mne-tools/mne-python/blob/main/mne/io/brainvision/brainvision.py#L146) uses `np.fromfile()`, which ignores endianness, but I think this function is never used (because the case `order == 'C'` is [not supported](https://github.com/mne-tools/mne-python/blob/main/mne/io/brainvision/brainvision.py#L83-L84) even though it seems to be implemented).

I think the other case `order == 'F'` is implemented [here](https://github.com/mne-tools/mne-python/blob/main/mne/io/brainvision/brainvision.py#L123-L131), I'm not sure if this enforces little endian in any way.

sappelhoff, 2021-09-21:
I would say, let's ensure we write in little-endian without BOM in pybv (that's the only thing we do), and go back to MNE and make sure that it reads all BV files as little-endian without BOM.

When all of this is clear, big endian systems should not have any problems

Aniket-Pradhan, 2021-09-23:
I made [some changes](https://github.com/Aniket-Pradhan/pybv/commit/d282e6d533fcd8778d1632f3e0100b5d31140bdf) to pybv so that it writes exclusively in the little-endian architecture.

The [travis build for s390x](https://app.travis-ci.com/github/Aniket-Pradhan/pybv/jobs/539196521) is successfully passing now.

This suggests that `mne` was indeed reading the files in little-endian format. I am quite unsure on how we can debug this, because when `mne` reads the files it [shifts the bytes](https://github.com/mne-tools/mne-python/blob/main/mne/io/brainvision/brainvision.py#L125) by setting offsets, and decodes the file [using ASCII](https://github.com/mne-tools/mne-python/blob/main/mne/io/brainvision/brainvision.py#L128) (which is endian-independent).

cbrnr, 2021-09-23:
So this means we currently don't understand why MNE is apparently always reading in little endian, right? We still cannot explain why the test file written on a big endian machine could not be read on the same machine. @larsoner do you know why MNE seems to enforce little endian in its BrainVision reader?

larsoner, 2021-09-23:
What does the file format say? IIRC usually it should be written one way or the other on any machine, not written according to whatever the native endianness is. To me it sounds like a bug with the writer that it writes in native endianness rather than always writing one of the two (probably little).

cbrnr, 2021-09-23:
Yes, the writer so far used the host endianness – but when we tested this on a big endian machine some of the tests failed. This should not have happened when both the writer and the reader used native endianness, so clearly either the reader or the writer must have assumed little endian somewhere. We're pretty sure the writer uses host endianness, so we were wondering if the MNE reader somehow assumes little endianness.

larsoner, 2021-09-23:
> We're pretty sure the writer uses host endianness, so we were wondering if the MNE reader somehow assumes little endianness.

Yes it probably does and I think this is the right thing to do. AFAIK you shouldn't use "host endianness" when writing a binary in a format, it means that you need to know the endianness of the host used to write a file. The right/standard thing to do I think is for the writer and reader to enforce a single endianness according to the file spec. Given that MNE can read native Brainvision files, little is probably what should always be read and also always written.

larsoner, 2021-09-23:
> We're pretty sure the writer uses host endianness

Wait when you say this, do you mean `pybv` uses native, or BrainVision's own software uses native? We should follow whatever BrainVision does and says should be done according to their spec. But it's also possible they never thought about writing on big endian architectures... might be worth asking them explicitly on some forum of theirs what the right thing to do is here.

cbrnr, 2021-09-23:
`pybv` did this. I think BrainProducts never thought about this use case, but they said their binaries should always be little endian (and they will add this to the specs). So yes, we will fix `pybv`, but can you explain how MNE already assumes little endian? I don't see this in the code.

larsoner, 2021-09-23:
> can you explain how MNE already assumes little endian? I don't see this in the code.

I am not super familiar with that code, but based on a quick look, _read_segments_file uses dtype_dict that specifies endianness

https://github.com/mne-tools/mne-python/blob/main/mne/io/brainvision/brainvision.py#L303

the `_c` variant looks like indeed it would be buggy/broken/incorrect since it does use `fromfile`. It's possible our tests would not catch this if they don't actually inspect the sanity of the data values...

cbrnr, 2021-09-24:
> I am not super familiar with that code, but based on a quick look, _read_segments_file uses dtype_dict that specifies endianness

Ah, of course! I didn't notice the `<` characters! That clears everything up, so all we need to do is enforce little endian in our writer.

> the _c variant looks like indeed it would be buggy/broken/incorrect since it does use fromfile. It's possible our tests would not catch this if they don't actually inspect the sanity of the data values...

It looks like that function is never called though: https://github.com/mne-tools/mne-python/blob/main/mne/io/brainvision/brainvision.py#L83-L84

sappelhoff, 2021-09-24:
@Aniket-Pradhan a pr to enforce little endian writing in pybv is welcome.

Re: Mne --> nice, so maybe we should just prune the unused function and add a small documentation somewhere that all files are read as little endian

That should have us covered then. Not sure if we should add a ci step to the pybv repo to run tests on a big endian machine. WDYT?

cbrnr, 2021-09-24:
> That should have us covered then. Not sure if we should add a ci step to the pybv repo to run tests on a big endian machine. WDYT?

I don't think that's necessary now that we have solved this issue (and tested it once).

Aniket-Pradhan, 2021-09-24:
> @Aniket-Pradhan a pr to enforce little endian writing in pybv is welcome.

Created [the PR](https://github.com/bids-standard/pybv/pull/80). I have also added in a small line in `docs/index.rst` about this *little* change. Let me know if there are any issues with it 😸 

> > That should have us covered then. Not sure if we should add a ci step to the pybv repo to run tests on a big endian machine. WDYT?
> 
> I don't think that's necessary now that we have solved this issue (and tested it once).

I agree with @cbrnr . There doesn't seem to be any other issues regarding endianness now. Either way, if there are any issues, it would be reported through Fedora's build system and I would be happy to report it here 😛

PR #80 (2021-09-29): Force little-endian format when writing eeg datafile
Also updated the docs to add information about this change 😸 

closes #79
closed 2021-09-29 by cbrnr