bids-standard/pybv #83: Add tests (CI) running Matlab/Octave for popular BV readers: FieldTrip, EEGLAB, Brainstorm

sappelhoff, 2022-05-22:
Having such a CI can help us ensure that what we write is not only read by MNE-Python, but also the other important packages.

This would be especially nice to have before addressing things like #77 #44

cbrnr, 2022-05-22:
Is there something official from Brain Products that's open source? I'd also consider BioSig (if it has a BV reader, not sure).

sappelhoff, 2022-05-22:
No, they only have the proprietary brain vision analyzer, which needs a physical license dongle (at least somewhere on the network). 

 +1 for biosig, if many people use it

cbrnr, 2022-05-23:
> +1 for biosig, if many people use it

No idea if many people still use it, but is has always been a reader with support for a lot of formats. If it is not too much effort I'd add it to the test matrix just to get another "opinion". There are Python bindings available, so it's probably easiest to use that instead of the C library directly (or the MATLAB version, which uses a different codebase).