bids-standard/pybv #93: allow events descriptions to contain 0 values

compmonks, 2022-06-30:
I have an event file with 'labels' (description) which contain 0 values. 
In this case, the labels are rotations in degrees but `write_brainvision ` fails during `_chk_events` and returns the following error: `events: when `type` is Stimulus, descriptions must be positve ints.`

Would it be possible to allow `event["description"]` to be at least >= 0 instead of >=1? The current test is 
```
443 if event["description"] < 1:
444                 raise ValueError(f"events: when `type` is {event['type']}, "
445                                  "descriptions must be positve ints.")```

compmonks, 2022-07-04:
@sappelhoff , thanks! I just created one there: 7afe170aebbdd7b23462c64866599577fb83b833.

sappelhoff, 2022-07-05:
closed in #95

closed 2022-07-05 by sappelhoff