bids-standard/pybv #110: Negative Event Onsets

behinger, 2023-11-20:
Hi!
According to BIDS standard, we can have negative onsets of events - but this runs into an assert here:
https://github.com/bids-standard/pybv/blob/cb79575c58ea898058177f2cfcc98dcbbebd9875/pybv/io.py#L471

is this a fundamental issue to be compatible with the BrainVision format, or is this something we could fix with a PR?

BIDS-Specificatoins:
[image]

Cheers and thanks for the hard work!

Best, Benedikt & Manpa (@reboreexplore)

sappelhoff, 2023-11-20:
Hello 👋 

> is this a fundamental issue to be compatible with the BrainVision format

Yes, I think it's a format issue 🤔 

But I also don't understand the use case --> in the BrainVision files, you'd store ephys data, and the VHDR file contains event markers into that data (the stored data). Do you want to store event markers in the VHDR that do not correspond to the stored data (EEG file) in any way?

behinger, 2023-11-20:
I see. We will switch to EDF then.

We use read_raw_xdf in mnelab to get a raws with negative annotations (lsl stimulus stream might be opened before the EEG data is available). We then use mne-bids to save that raw file. where exactly in mne-bids pybv comes into play I'd have to look into.

sappelhoff, 2023-11-20:
Okay thanks, that makes sense to me -- and switching to EDF sounds like a reasonable alternative then! Good luck!

closed 2023-11-20 by behinger

cbrnr, 2023-11-20:
> We use read_raw_xdf in mnelab to get a raws with negative annotations (lsl stimulus/marker stream might be opened before the EEG data is available).

Just out of curiosity, shouldn't the streams be synced once you load it with `read_raw_xdf()`?

behinger, 2023-11-20:
I thought so too, but for some reason we provide only the EEG stream, somehow it has to get the Marker streams without providing explicitly  - not 100% sure yet. 

so we call mnelab with: https://github.com/s-ccs/LSLAutoBIDS/blob/b055f107355fffdc94447f9b5524573cb50409b4/lsl_autobids/convert_to_bids_and_upload.py#L135

Which should be one stream - but now looking at this code a second time, I'm not so sure anymore. @reboreexplore could you put a breakpoint there and check what exactly is inputted to read_raw_xdf? 

But, because this is off-topic, we can take the discussion offline and we'll provide a summary of our findings later :-)

ReboreExplore, 2023-11-20:
Yes, I could see into this matter.