bids-standard/pybv #119: Why is `_export_mne_raw()` private?

cbrnr, 2024-02-19:
Is there a reason why `_export_mne_raw()` is a private function? It's only used in some tests, but it would probably be useful for some people (like me 😄).

sappelhoff, 2024-02-19:
It's because exporting should be done from MNE-Python, using https://mne.tools/stable/generated/mne.export.export_raw.html#mne.export.export_raw

this way we only really have to maintain one API, instead of two. And people who want to export from mne to brainvision would have both dependencies (mne, pybv) anyhow.

Does that make sense? I'm happy to discuss alternatives.

sappelhoff, 2024-02-19:
actually, we probably should have a small section "exporting from mne-python" right about here: https://github.com/bids-standard/pybv?tab=readme-ov-file#writing-brainvision-files

PR #120 (2024-04-21): Remove MNE export
MNE export should be handled by MNE, so I'm removing related functions and tests (and will be including them in MNE in a separate PR). Fixes #119.
closed 2024-04-21 by sappelhoff